beagleboard/beaglebone-ai #2: SW: need Tensorflow/Python integration with TIDL

jadonk, 2019-06-07:


jadonk, 2019-09-17:
Tensorflow Lite support is on the roadmap for 2020Q1.

PrashantDandriyal, 2020-03-18:
With the release of [TIDL 3.15.1](http://downloads.ti.com/processor-sdk-linux/esd/docs/latest/linux/Foundational_Components/Machine_Learning/tidl.html#simulation-tool), TensorFlow and TensorFlow Lite has been added to the _supported frameworks_ among _Caffe, TensorFlow Lite and ONNX_ although, **not yet** a verified model like the _Jacinto11, MobileNetV2 ,etc_.

The _model import_ feature of TIDL there is provides constraint on the supported operators and layers. Irrespective of the framework, the supported layers are 
```Convolution Layer
Pooling Layer (Average and Max Pooling)
ReLU Layer
Element Wise Layer (Add, Max, Product)
Inner Product Layer (Fully Connected Layer)
Soft Max Layer
Bias Layer
Deconvolution Layer
Concatenate layer
ArgMax Layer
Scale Layer
PReLU Layer
Batch Normalization layer
ReLU6 Layer
Crop layer
Slice layer
Flatten layer
Split Layer
Detection Output Layer
```

But for the supported operators, during the import process, some operators/layers are combined or converted into TIDL layers. For TensorFlow specifically, the supported operators are:

TensorFlow Operator | TIDL Layer
-- | --
Placeholder | TIDL_DataLayer
Conv2D | TIDL_ConvolutionLayer
DepthwiseConv2dNative | TIDL_ConvolutionLayer
BiasAdd | TIDL_BiasLayer
Add | TIDL_EltWiseLayer
Mul | TIDL_ScaleLayer
FusedBatchNorm | TIDL_BatchNormLayer
Relu | TIDL_ReLULayer
Relu6 | TIDL_ReLULayer
MaxPool | TIDL_PoolingLayer
AvgPool | TIDL_PoolingLayer
ConcatV2 | TIDL_ConcatLayer
Slice | TIDL_SliceLayer
Squeeze | See note below
Reshape | See note below
Softmax | TIDL_SoftMaxLayer
Pad | TIDL_PadLayer
Mean | TIDL_PoolingLayer

Whereas for TensorFlow Lite, 
TensorFlow Operator	|TIDL Layer
--|--
Placeholder|	TIDL_DataLayer
CONV_2D	|TIDL_ConvolutionLayer
TRANSPOSE_CONV	|TIDL_Deconv2DLayer
DEPTHWISE_CONV_2D	|TIDL_ConvolutionLayer
ADD	|TIDL_EltWiseLayer
MUL	|TIDL_ScaleLayer
RELU	|TIDL_ReLULayer
RELU6	|TIDL_ReLULayer
MAX_POOL_2D	|TIDL_PoolingLayer
AVERAGE_POOL_2D	|TIDL_PoolingLayer
CONCATENATION	|TIDL_ConcatLayer
RESHAPE	|See note below
SOFTMAX	|TIDL_SoftMaxLayer
ARG_MAX	|TIDL_ArgMaxLayer
PAD	|TIDL_PadLayer
MEAN	|TIDL_PoolingLayer
FULLY_CONNECTED	|TIDL_InnerProductLayer

Note:
>“Reshape” and “Squeeze” are supported by being coalesced into other layers:
If “Reshape” immediately follows “Squeeze”, they both are coalesced into TIDL_FlattenLayer.
If “Reshape” immediately follows “AvgPool”, “Reshape” is coalesced into TIDL_PoolingLayer.
If “Reshape” immediately follows TIDL_InnerProductLayer, it is coalesced into TIDL_InnerProductLayer.

Source: [Official docs for TIDL: Foundational Components ](http://downloads.ti.com/processor-sdk-linux/esd/docs/latest/linux/Foundational_Components/Machine_Learning/tidl.html#simulation-tool)

matthewanderson23, 2020-04-28:
Any updates on whether TF Lite support is imminent?