beagleboard/beaglebone-ai #3: SW: need mainline WiFi/BT driver support

jadonk, 2019-06-07:
Should also include 4.14-ti/4.19-ti backports.

vlasov01, 2019-11-16:
I'd like to connect BBAI to connect to a BT speaker.
I'm wondering if there are any workarounds to use BT to accomplish this?

`sudo bluetoothctl
[bluetooth]# show
No default controller available
`

`sudo dmesg | egrep -i 'blue|firm'
[   50.217975] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Mar  1 2015 07:29:38 version 7.45.18 (r538002) FWID 01-6a2c8ad4
[   51.907105] Bluetooth: Core ver 2.22
[   51.907296] Bluetooth: HCI device and connection manager initialized
[   51.907335] Bluetooth: HCI socket layer initialized
[   51.907369] Bluetooth: L2CAP socket layer initialized
[   51.907460] Bluetooth: SCO socket layer initialized
[   51.940582] Bluetooth: Generic Bluetooth SDIO driver ver 0.1
[   63.455873] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   63.455897] Bluetooth: BNEP filters: protocol multicast
[   63.455948] Bluetooth: BNEP socket layer initialized

`

nullr0ute, 2020-08-16:
Is the broadcom NVRAM file published somewhere?

sPyOpenSource, 2021-02-14:
I have the same problem with Bluetooth.

sPyOpenSource, 2021-02-14:
You need to compile and install overlay: https://github.com/beagleboard/BeagleBoard-DeviceTrees/blob/v4.19.x-ti-overlays/src/arm/am5729-beagleboneai.dts
Then download https://github.com/RPi-Distro/bluez-firmware/blob/master/broadcom/BCM4345C0.hcd put into /etc/firmware
```
hciattach /dev/ttyS5 bcm43xx 921600
sleep 3
hciattach /dev/ttyS5 bcm43xx 921600
```
Then after reboot everytime using 
`btattach -B /de/ttyS5 -S 921600`