beagleboard/beaglebone-ai #7: Doc:  The SRM pin tables are a bit hard to navigate

MarkAYoder, 2019-07-09:
I've copied them to some Google Sheets to aid in navigation.
PRU Pins: https://docs.google.com/spreadsheets/d/19u2rtqVGChSOd0_8ZJj_E3XosuGBpWy7j633__siOvw/edit?usp=sharing
P8 Pins: https://docs.google.com/spreadsheets/d/1opaOLjF7_EDNCR3g5NiEILsJba8h5ZumSeBwOvD1-6Q/edit?usp=sharing
P9 Pins: https://docs.google.com/spreadsheets/d/12qkmnyM3qgxL-F6tBKBaA220u_hGWrLgVg9rrCykZ2Q/edit?usp=sharing

I can add them to the Wiki and to a pull request if you'd like.

--Mark

jadonk, 2019-07-14:
@MarkAYoder yes, please submit your improvements. @cwicks is working on the SRM with some help from @JohnThanks and @ErikWelsh.

MarkAYoder, 2019-07-16:
@cathywicks I don't see how to submit improvements to the Wiki.  I found the paths above easier to use than the tables in the wiki.  Maybe you could add the links to the Wiki.

--Mark

jadonk, 2019-07-17:
@MarkAYoder there should be an edit button you can use on the wiki pages.

cathywicks, 2019-07-17:
Hi Mark. I’ve heard we should go ahead and put the pin tables in a
scrolling chart. I’ll work on that by the weekend. I would love to link to
any chart folks have that is printable. Do you have one?

MarkAYoder, 2019-07-17:
Cathy:
  I've copied the pin mux tables to Google Sheets.  They appear to be nicely printable.

PRU Pins: https://docs.google.com/spreadsheets/d/19u2rtqVGChSOd0_8ZJj_E3XosuGBpWy7j633__siOvw/edit?usp=sharing
P8 Pins: https://docs.google.com/spreadsheets/d/1opaOLjF7_EDNCR3g5NiEILsJba8h5ZumSeBwOvD1-6Q/edit?usp=sharing
P9 Pins: https://docs.google.com/spreadsheets/d/12qkmnyM3qgxL-F6tBKBaA220u_hGWrLgVg9rrCykZ2Q/edit?usp=sharing

The PRU pin table is missing the pr2 entries.

I don't know how the edit the wiki, so it looks like you'll have to include the links.

--Mark

cathywicks, 2019-07-17:
thanks Mark!  I can include the links on the SRM

jadonk, 2019-09-18:
@cathywicks adding the links would be good. I've tried to make a "master sheet" at https://docs.google.com/spreadsheets/d/1fE-AsDZvJ-bBwzNBj1_sPDrutvEvsmARqFwvbw_HkrE/edit?usp=sharing 

@MarkAYoder can you review the section 7 and section 8 header pin information?

MarkAYoder, 2019-09-20:
I see the spreadsheet, but I don't what is section 7 and 8.

silver2row, 2021-02-22:
Hello...please forgive me bringing up old times here but are the links good for P8 and P9 for the AI?

Seth