beagleboard/beaglebone-ai #10: Problems with input power muxing using Schottky ORing diodes?

jks-prv, 2019-09-08:
Thanks for recently adding the rev A1 design files, especially the schematic.

Are you guys sure about using Schottky ORing diodes (D81,82) for input multiplexing between the USB-C and cape power-in sources? There are a couple problems I can see.

Depending on current draw by the AI you might be exceeding the power dissipation of the diodes. Their Pd is spec'd at 850 mW and you'll hit that by putting only ~1.5A through them
(half of what the USB-C input is capable of). The BBB avoided this of course by using low-loss FET muxing inside the PMIC.

I ran some simulations of your Q15 PFET + B360A diode circuit and over the USB-C 5V power adapter voltage supply range spec of 4.75 - 5.5V you're not going to meet the VDD Vmin input spec on your USB-C CCL (TUSB322I U22). It's short about 0.2V depending on the Vf of the diodes. [Simulation results](http://kiwisdr.com/files/BBAI.2A.png) with AI drawing ~2A.

If it's late in the game to be making changes you might at least want to see if you can spec a better Schottky with lower Vf (e.g. SSB44). A much better solution would be to use something like a TPS2120 in place of the diodes.

jadonk, 2019-09-17:
Cape expansion only ever specified up to 500mA over SYS_VDD.

[image]

jadonk, 2019-09-18:
Providing a bit more margin would be a good update for rev A2.

jks-prv, 2019-09-20:
First, congrats on shipping rev A1 yesterday. I've ordered one and will test with my cape ([kiwisdr.com](http://kiwisdr.com)) which powers the BBB/G via P9-5,6 and is in turn fed with switched 5V from the PMIC via P9-7,8. So these issues are very important for my product if the AI is to be an effective performance upgrade.

> Cape expansion only ever specified up to 500mA over SYS_VDD [really SYS_5V]

Agreed, but that's not what I'm talking about. Please refer to this picture:
[image]
Powering the BBB from the cape via P9-5,6 (as an alternative to the DC-in jack) is allowed per SRM 8.6.2 "Expansion board external power". Those pins together allow 2A, so no problem powering the BBB plus 500mA of sequenced (switched) cape power on P9-7,8 going back out again. I use this arrangement because the DC-in jack on my cape has additional filtering compared to the DC-in jack on the BBB. But I still need power to my board to be sequenced by the PMIC so as not to damage the BBB. Hence I use P9-5,6 as BBB power-in and P9-7,8 as cape power-in. The AI supports this configuration so no problem.

(1) (2) As mentioned in the first post the D81,82 Schottky diodes may have heat dissipation and excessive voltage drop problems (relative to VDD Vmin spec of U22) depending on the total 5V current draw and Vf of the diodes. The PMIC, U48 and other downstream users of VDD_5V (including my cape) will have no problem with the diode Vdrop because they are all LDOs with low input Vmin. Points (1) and (2) are my main area of concern.

(3) This issue is just for completeness. I'd be surprised if it makes any difference to anyone. The BBB SRM allows for P9-5,6 to be bidirectional, i.e. if powered from the DC-in jack the cape can receive power on P9-5,6 (PMIC unswitched) in addition to P9-7,8 (switched). The AI precludes this configuration due to the unidirectional D82. Replacing D81,82 with MOSFET switching would solve this problem.

jadonk, 2019-09-26:
Can you propose a circuit?

jadonk, 2019-09-26:
Also, your pictures are fantastic. Can I use them in the documentation?

jks-prv, 2019-09-26:
I'm going to dead bug a TPS2120 in place of D81,82 and try it out. In Q9k on a reel from Mouser the part is US $0.86 which is a little pricey. But you can deduct the cost of the diodes from that. And also I'm sure you guys get better than distribution pricing from TI.

Sure, use/modify the pics as you wish. A .pdf and the OpenOffice .odg source are on http://kiwisdr.com/files/Schottky.[pdf,odg]

embedded-solutions, 2019-11-07:
I want to contribute to discussion with some measurements. I performed some simple tests to see the voltage drop on diode by measuring voltage on pins P9 7,8pins. 

On the plot we can see complete boot sequence and voltage drop

1 - BBAI powered from lab power supply without load on pins 7,8
 (black - voltage on pins 5,6 | green - voltage on pins 7,8)
[image]

2 - BBAI powered from lab power supply with 500mA load on pins 7,8
[image]

3 - BBAI powered from USB C supply with 500mA load on pins 7,8
[image]

In the last case board does not boot at all.

buzmeg, 2020-05-07:
This should really have something like an "ideal diode" or a dual ideal diode like the LTC4413 (I'm sure TI has an equivalent somewhere) if there is no way to do this inside the PMIC.

jadonk, 2020-05-14:
[image]

buzmeg, 2020-06-27:
Seems like a lot of components just for an ideal diode, but, yeah, that should do.

However, I don't think you want to enable PRI mode.  You probably want simple "Pick the higher voltage" mode.

With the PR1 configuration, you can have unwanted switching around 4V if the TypeC_VBUS is struggling.  With "higher voltage" mode, the switching point is set by the stronger/higher of the two voltages and likely isn't at 4V.

For example, the system is drawing power from TypeC_VBUS which sinks below 4V because it can't keep up.  The TPS2121 switches over to VDD_CAPE_5V which releases the load on TypeC_VBUS.  At that point, TypeC_VBUS comes back up to above 4V and the TPS2121 switches back to it.  At which point, the problem starts again.