beagleboard/beaglebone-ai #17: SW: Add SGX acceleration to Debian images

jadonk, 2019-09-17:
cc @RobertCNelson

jadonk, 2019-09-17:
Right now, you can only use EGL for 3D acceleration, such as with with a fresh build of Qt (ie., not the Debian-built Qt which is Xorg only).

silentjet, 2020-02-08:
@jadonk is there SGX SDK available to build accelerated graphics for my DIY linux?

jadonk, 2020-05-15:
@RobertCNelson is there something to escalate to TI on this?

RobertCNelson, 2020-05-15:
Default Kernel has the SGX Kernel modules installed..

User Mode Libraries are available: https://git.ti.com/gitweb/?p=graphics/omap5-sgx-ddk-um-linux.git;a=summary

Someone just needs to test them..

jadonk, 2020-05-16:
@silentjet Can you test them?

silentjet, 2020-05-18:
> @silentjet Can you test them?

Perfect timing, just ordered board. (Going to port my solution from black to ai)