beagleboard/beaglebone-ai #21: HW: wire mods required to enable JTAG

jadonk, 2019-09-18:
Got this message forwarded to me today about the board's JTAG:

Hi Bill

I finally manage how to use the Allegro Board Viewer.

It’s confirmed that the JTAG pins are shorted together. I’m not sure if they even connected to the SOC.

[image]

It looks like R199 and R200 are just connected together in this photo above. SOC -> R200 -> R199 -> SOC ( the red trace that has the letter JTAG_RTCK on it)

And the pins 9-11 on the JTAG connectors are directly connected to each other

[image]

So we can cut this trace, and the trace between R199 and R200, then blue wires to the R200 and R199. We only have to do 1 blue wire.

That will probably fixe it :-)

Best regards
Thanh

sgturner, 2019-10-03:
I someone able to pinpoint where R199 and R200 are on the PCB (since there are no R and C references on the silkscreen)?

dar-jan, 2019-10-03:
Stephen, you've right. PCB silkscreen without Names is hard to recognise. Resistors are near the sdcard corner (R199 at the corner and R200 first outside), see: 
[image]
The pdf with PCB from documentation  (https://github.com/beagleboard/beaglebone-ai/blob/master/BeagleBone-AI_brd.pdf) presents only top side, without bottom...

sgturner, 2019-10-03:
Hi Dariusz,

Can you pinpoint on the below image?.  Thanks

[image]

dar-jan, 2019-10-03:
[image]
Pure uP JTAG signals appear on right side of resistors at your picture.

sgturner, 2019-10-07:
I believe I have JTAG working on the BB-AI.

1. I cut the trace joining JTAG_TCK and JTAG_RTCK here:

[image]

2. Unsoldered and resoldered R199 and R200.  Wired R200 to the JTAG header pin 11 (TCK).  Did not wire RTCK as it is not used by the XDS110 JTAG debugger that I'm using.  Soldered a 20 pin header (with pin 6 removed) to P2

[image]

```
[Start: Texas Instruments XDS110 USB Debug Probe]

Execute the command:

%ccs_base%/common/uscif/dbgjtag -f %boarddatafile% -rv -o -S integrity

[Result]

-----[Print the board config pathname(s)]------------------------------------

C:\Users\AUDIOS~1\AppData\Local\TEXASI~1\
    CCS\ccs910\0\0\BrdDat\testBoard.dat

-----[Print the reset-command software log-file]-----------------------------

This utility has selected a 100- or 510-class product.
This utility will load the adapter 'jioxds110.dll'.
The library build date was 'Aug 26 2019'.
The library build time was '13:34:49'.
The library package version is '8.3.0.00003'.
The library component version is '35.35.0.0'.
The controller does not use a programmable FPGA.
The controller has a version number of '5' (0x00000005).
The controller has an insertion length of '0' (0x00000000).
This utility will attempt to reset the controller.
This utility has successfully reset the controller.

-----[Print the reset-command hardware log-file]-----------------------------

The scan-path will be reset by toggling the JTAG TRST signal.
The controller is the XDS110 with USB interface.
The link from controller to target is direct (without cable).
The software is configured for XDS110 features.
The controller cannot monitor the value on the EMU[0] pin.
The controller cannot monitor the value on the EMU[1] pin.
The controller cannot control the timing on output pins.
The controller cannot control the timing on input pins.
The scan-path link-delay has been set to exactly '0' (0x0000).

-----[Perform the Integrity scan-test on the JTAG IR]------------------------

This test will use blocks of 64 32-bit words.
This test will be applied just once.

Do a test using 0xFFFFFFFF.
Scan tests: 1, skipped: 0, failed: 0
Do a test using 0x00000000.
Scan tests: 2, skipped: 0, failed: 0
Do a test using 0xFE03E0E2.
Scan tests: 3, skipped: 0, failed: 0
Do a test using 0x01FC1F1D.
Scan tests: 4, skipped: 0, failed: 0
Do a test using 0x5533CCAA.
Scan tests: 5, skipped: 0, failed: 0
Do a test using 0xAACC3355.
Scan tests: 6, skipped: 0, failed: 0
All of the values were scanned correctly.

The JTAG IR Integrity scan-test has succeeded.

-----[Perform the Integrity scan-test on the JTAG DR]------------------------

This test will use blocks of 64 32-bit words.
This test will be applied just once.

Do a test using 0xFFFFFFFF.
Scan tests: 1, skipped: 0, failed: 0
Do a test using 0x00000000.
Scan tests: 2, skipped: 0, failed: 0
Do a test using 0xFE03E0E2.
Scan tests: 3, skipped: 0, failed: 0
Do a test using 0x01FC1F1D.
Scan tests: 4, skipped: 0, failed: 0
Do a test using 0x5533CCAA.
Scan tests: 5, skipped: 0, failed: 0
Do a test using 0xAACC3355.
Scan tests: 6, skipped: 0, failed: 0
All of the values were scanned correctly.

The JTAG DR Integrity scan-test has succeeded.

[End: Texas Instruments XDS110 USB Debug Probe]
```

Next step is to find a GEL file for the BB-AI.  I note the latest release of TI Code Composer (v9.1.0) does not even have an AM5729 target

mgsb, 2019-11-06:
I would like to try this, too. Would you clarify for me what you did with R199 and R200? I don't understand what you mean by: "Unsoldered and resoldered R199 and R200."?

sgturner, 2019-11-06:
You can see what I did here:

[image]

I unsoldered R199 and R200 and then resoldered them to the left hand pad.  Only connected up R200 (TCK)

EnzoS1469, 2019-12-12:
Hi,
are above hw modifications valid for XDS560v2 JTAG debugger too ? 
Specifically, I read that RTCK is not wired because  XDS110 JTAG debugger is not using it: do I need to wire in case of  XDS560v2 JTAG debugger ?

Last question: what are the impacts of these modifications ? Is any functionality limited/dropped ? 

Thank you in advance,
Enzo

rsjdesouza, 2020-05-05:
Enzo,

I was able to work with my Blackhawk XDS560v2 but tied both signals so I could potentially achieve more stability and higher speeds. By using the TCK in the rising edge, I was able to reach up to 26MHz of TCK. 

[BBAI_XDS560v2_test.txt](https://github.com/beagleboard/beaglebone-ai/files/4582007/BBAI_XDS560v2_test.txt)

I followed sgturner's procedure above with a slight difference. 

As recommended by the section _Non-buffered JTAG Signal Termination_ of the [XDS Target Connection Design](https://software-dl.ti.com/ccs/esd/documents/xdsdebugprobes/emu_xds_target_connection_guide.html), I moved the 22Ω termination resistor (R200) of the TCK signal to be close to the emulation header, while I left the 22 (R199) of the RTCK as near the device as possible. I used very thin _wire wrap_ wire and some kapton tape to keep things in place. 

Also, it is possible to cut the connection between pins 9 (RTCK) and 11 (TCK) of the emulation header after it was installed (that was my case). I used a small and very thin exacto knife to scrape between the pads of the header. Since the connection is in the bottom layer, I was very careful to not go too deep as I was afraid to cut other tracks on inner layers (I didn't verify this).

A picture of the modifications is shown below. 

[image]

jadonk, 2020-05-14:
Rev A1/A1a:
[image]

Rev A2:
[image]

udayshukla14, 2020-10-22:
Hello,
I have recently purchased BB AI board. I have done the modifications as per the discussions on this page to support spectrum debugger.
After modifications, as soon as I connect the debugger adapter to the 20 pin connector on BB AI, it turns off automatically.
Has anyone faced similar issues ?
Please note, I have retained all 20 pins intact on the JTAG connector.

jostmp, 2022-05-20:
@udayshukla14 did you resolve the power off issue? I am seeing the same thing when I attach an XDS110 adapter to the board.

udayshukla14, 2022-05-21:
Sorry @jostmp , unfortunately I had to drop the project on BB since I could not resolve the issue

jostmp, 2022-05-27:
I found that with my XDS110 I could not use the 20 pin TI adapter. As soon as I connected to the board it powered down. Removing pin 6 as noted by @sgturner and using a 20 pin 50mil ARM debug cable connected directly to the board and XDS110 all works well.