beagleboard/beaglebone-ai #24: HW: need pull-down on console uart RX line

jadonk, 2019-09-18:
@embest-tech said:

> We found the board goes to uboot mode when the board start without serial cables, the reason is that U46 2# pin input is floating , even if we had a pull-up resistor(R173) in the side of cpu, it will not be effective. BeagleBone AI Rev A1 increases the length of the UART1_RX signal by moving the position of the serial port, this line is very sensitive without pullup and pulldown resistor , When we touch the edge of the PCB, it brings in a lot of interference signals, causing the serial port to trigger into UBOOT mode. we suggest add a pull-down resistor on the UART1_RX signal, is it feasible to modify it in the same way as Beaglebone Black?We have tested pull-down resistor to J92 2# , this problem has not been reproduced. We suggest add a small pull-down resistor between J92 1# and J92 2# in this PP production.
> 
> [image]
> 

In assembly of rev A1a, R173 is removed and a pull-down is added on J92 pin 2.

For rev A2, the PCB will be modified to avoid the need for this unusual assembly step.

arashed31, 2019-09-20:
Can you try this with a pull up resistor on both side of U46? The standard is to keep a logic high on UART RX to indicate RS232 idle state.

VladAugust, 2019-10-10:
J92 pin 2 should be pulled up to VCC because 

>  The standard is to keep a logic high on UART RX to indicate RS232 idle state.

sharifsonic, 2020-01-11:
hi,
how can we boot BB AI from sd card slot?
didnt find any instructions

DCelectronics, 2020-01-31:
> 
> 
> hi,
> how can we boot BB AI from sd card slot?
> didnt find any instructions

I am not sure if the question relates to uart RX issue.
BeagleBone AI should automatically boot from the microSD once a valid OS image is found on the microSD.
Using the recommended Debian image, which at the time of this writing is found at:
https://debian.beagleboard.org/images/am57xx-debian-9.9-lxqt-armhf-2019-08-03-4gb.img.xz
on a quality microSD should work "right out of the burner".
If you are booting on microSD and you want to switch back to eMMC, perform a complete poweroff, remove the microSD from its holder, and power up the BeagleBone AI.
There is no BOOT button like the one found on BeagleBone Black.

sharifsonic, 2020-02-01:
> I am not sure if the question relates to uart RX issue.
> BeagleBone AI should automatically boot from the microSD once a valid OS image is found on the microSD.
> Using the recommended Debian image, [...]

Thanks, sorry for posting in the wrong place

DCelectronics, 2020-02-02:
> Thanks, sorry for posting in the wrong place

No worries, you have to get started from somewhere. 
I would much rather you get your question answered.

BerndMoe, 2020-03-05:
Hello,

I have a new BB AI Rev. A1 and it shutdown after overheat. Now I search for solving of the problem and I found this thread.
The head of this thread is "need pull-down". @embest-tech write that need a pull up, that is correct. When you connect the Rx with a resistor to Tx, so the Rx side receive the Tx without connection to a UART terminal. So you have design a hardware UART loop.
The most picture in the web and my BB AI have a resistor between pin 1 and 2, so we have a pull-down. Without UART connection exists no problems. Only when you after booting connect a UART the Rx side receive a failure byte automatically.

Have fun.

anvesh-design, 2020-03-06:
Hi,
i am new to beagle bone. i would like to know why there is a SN74LVC2G241 at UART0? why can't direct connection to Tx and Rx to header pins.

thank you,
anvesh

arashed31, 2020-03-06:
The IO pins on the processor can get damaged if you try to drive the pins
from an external device while the processor is off. So the buffer IC is
used to prevent that from occurring.

DCelectronics, 2020-03-15:
100% agree with "The IO pins on the processor can get damaged if you try to drive the pins from an external device while the processor is off. So the buffer IC is used to prevent that from occurring."

I have a BB AI Rev. A1
There is a resistor between pins J92-1 and J92-2.
[image]
Also. the resistor R173 is installed (populated on the board). The R173 is a "weak" pull up but the SN74LVC2G241 buffer (which has low output resistance driver) will assert whatever signal floats on pin U46-2 (the input of the buffer). They put a "weak" pull down resistor to prevent the pin from floating, and the GND net was convenient (they didn't have to route 3V3 net).  
Beaglebone.org did the same thing for BB Black.

I have an FTDI UART cable, and I noticed that it has an internal pull-up RX resistor which addresses the concern regarding "The standard is to keep a logic high on UART RX to indicate RS232 idle state."

If you use an FTDI UART cable, I strongly recommand to plug the UART cable before powering  your BB and disconnect it after you power off your BB WHILE the UART cable is also connected to (and powered from) your host computer.

Regarding the overheating issue, I use a fan, and it is effective.

DCelectronics, 2020-03-17:
Suggested changes for revision A2 are:
Add a new 100K pull-up resistor (R?) 
[image]
Make R173 "NI" (not installed).
[image]
The 100K resistor R? can be added next to U46, where one can conveniently route the VDD_3V3 net to the resistor.
[image]

jadonk, 2020-05-14:
[image]

[image]