beagleboard/beaglebone-ai #26: Yocto support?

reneherrero, 2019-09-21:
Anyone know if there's a Yocto BSP layer for the Beaglebone AI?

All I found is this but it's very stale: https://github.com/beagleboard/meta-beagleboard

Thanks, and apologies if this is not the appropriate forum.

jadonk, 2019-09-26:
Good to check at https://beagleboard.org/discuss for faster answers, but this does maintain an open issues list. TI Processor SDK support will eventually get done. I expect around January.

jadonk, 2019-09-26:
oh, and it would be in meta-ti, not meta-beagleboard, unless Koen decided he wanted to jump back in and support.

reneherrero, 2019-09-28:
Appreciate the info @jadonk. And thanks for pointing me to the discuss page: never noticed it.

I propose leaving this open until a v1 comes out?  I had come across some of those googlegroups discussions via Google search results in the past, but never caught on as to how to tune in (GitHub & Slack is where I usually hang out).  I'm guessing I'm not alone...

reneherrero, 2019-12-09:
FYI: the term 'beaglebone ai' is starting to come out on the thud-next branch:

http://git.yoctoproject.org/cgit/cgit.cgi/meta-ti/log/?h=thud-next

Anyone know where the folks working on meta-ti collaborate? (if they do so publicly)

closed 2022-11-29 by reneherrero