beagleboard/beaglebone-ai #28: SW: default /dev/spidev1.0 spi2 mosi/miso reversed compared to BBB

jks-prv, 2019-09-30:
I'm still researching this issue (I don't know yet where to look to find the '5729 spi driver code). But I thought I better mention it now and possibly save someone some grief in these early days.

I have the AI spi2 working using my own dts derived from am5729-beagleboneai-roboticscape.dts.
When in master mode the BBB defines spi0 as:
```
P9.17 = spi0_cs0
P9.18 = spi0_d1(mosi)
P9.21 = spi0_d0(miso)
P9.22 = spi0_sclk
```
BBAI defines spi2 as:
```
P9.17a = spi2_cs0
P9.18a = spi2_d0
P9.21b = spi2_d1
P9.22b = spi2_sclk
```
Yes, d[01] are reversed w.r.t the P9 pins. But this is okay. There are new bits in the '5729 spi controller to assign the I/O role of d[01] to be whatever you want (i.e. it's no longer fixed like it was in the '3359).

The problem is that the default when you open /dev/spidev1.0 seems to be
```
P9.18a = spi2_d0(miso)
P9.21b = spi2_d1(mosi)
```
i.e. you get output on P9.21 which is the opposite of the BBB behavior (output on P9.18)
Everthing else about spi2 is fine: cs0 and sclk are correct, I can change speeds and word length etc.

I wrote a user-land program to mmap() the spi2 dev regs. For AM572x TRM rev/K the relevant pages are:
- 24.4.2.1 page 5930 describes that the spi2_d[01] I/O role is determined by MCSPI_CHxCONF bits IS, DPE1 and DPE0
- 24.4.4.1 page 5942 block diagram shows nicely how these bits control the interface pins
- Table 24-333 page 5980 describes the MCSPI_CHxCONF reg

The documented reg reset value from above is IS,DPE[01]=110 which is spi2_d[01]=mosi,miso which is correct and compatible with BBB. However before even opening /dev/spidev1.0 I observe IS,DPE[01]=001 which is the exact opposite (spi2_d[01]=miso,mosi). I can force the bits to =110, but after an open or ioctl on /dev/spidev1.0 the bits go back to =001. So driver software someplace is definitely setting them this way.

From show-pins.pl:
```
P9.22b                   240  A26 0 fast rx  up  spi2_sclk        kiwisdr (cape_pins_kiwi)
P9.21b                   241  B22 0 fast rx  up  spi2_d1          kiwisdr (cape_pins_kiwi)
P9.18a                   242  G17 0 fast rx  up  spi2_d0          kiwisdr (cape_pins_kiwi)
P9.17a                   243  B24 0 fast     up  spi2_cs0         kiwisdr (cape_pins_kiwi)

```

In the .dts for d[01] I've tried various combinations of PIN_INPUT and PIN_OUTPUT in the DRA7XX_CORE_IOPAD(). It doesn't make any difference. I was hoping the driver might pay attention to those in setting the MCSPI_CHxCONF bits.

jks-prv, 2019-10-01:
Found the solution. Simply add `ti,pindir-d0-out-d1-in` to your .dts file:
```
&mcspi2 {
	#address-cells = <1>;
	#size-cells = <0>;
	status = "okay";
	ti,pindir-d0-out-d1-in;
...
```

closed 2019-10-01 by jks-prv

Jayasudhaa, 2020-08-04:
Hi,

i am struggling to configuring spi on beagle bone AI.
how can i do it?

juvinski, 2021-01-05:
Thanks man,  I was strugling and loosing until I found your post :+1: 
:clinking_glasses: