beagleboard/beaglebone-ai #29: whats difference between beaglebone and tda2x?

DinghaoZhang, 2019-10-02:
hi, whats difference between beaglebone and tda2x?

DinghaoZhang, 2019-10-13:
The core of beaglebone ai and tda2x is the same, both are dual a15, 2 c66x and 4 eve. Maybe tda2x is also based on AM5729 (I guess). However, their peripheral circuits are not quit the same.

silentjet, 2020-02-08:
> The core of beaglebone ai and tda2x is the same, both are dual a15, 2 c66x and 4 eve. Maybe tda2x is also based on AM5729 (I guess). However, their peripheral circuits are not quit the same.

TDA2x is based on automotive soc Jacinto6 which was a derivation of omap4 approach -> omap3
BBAI is based on industrial soc Sitara family which is a derivation of the omap3 approach

Obviously they are reusing a lot of the same IP cores and solutions, but the ideas stand behind these two SoCs are bit different.

jadonk, 2020-04-09:
DRA7x, TDA2x and AM57x are all part of the same SoC family with various peripheral and processor options within them.

closed 2020-04-09 by jadonk