beagleboard/beaglebone-ai #34: Cortex-M4 sample code

reneherrero, 2019-12-09:
There's a lot of focus on the AI pieces of this board, but I find the 2x dual Cortex-M4 co-processors aren't getting the attention they deserve.

To anyone possibly sharing my sentiment, here's the closest I've been able to find:
* The Processor SDK for Linux: http://software-dl.ti.com/processor-sdk-linux/esd/AM57X/latest/index_FDS.html
* A demo project in Rust using the M4s and mailbox on the X15: https://github.com/cambridgeconsultants/rust-beagleboardx15-demo

Hope this helps

silver2row, 2020-01-13:
Hello,

Nice!

Seth

PrashantDandriyal, 2020-03-21:
The Processor SDK 3.15 for the AM572x can be found here : http://downloads.ti.com/processor-sdk-linux/esd/docs/latest/linux/Foundational_Components/Machine_Learning/tidl.html

 The following snippets help in highlighting the role of the A15 processors, although not in the form of a pseudocode or sample codes.

* In the software stack, as can be summarised in the figure:
[image]

In cases when user uses TIDL + DSPs or alternatively TIDL + EVEs or a combination of both (TIDL + DSPs + EVEs), the software layers used are: 
>In case TIDL uses DSP as accelerator there are three software layers:
> * TIDL Library that runs on DSP C66
>* OpenCL run-time, which runs on A15, and DSP
>* TIDL API host wrapper, user space library

>In case TIDL uses EVE as accelerator there are four software layers:
>* TIDL Library that runs on EVE
>* M4 service layer, acting as proxy between EVE and **A15** Linux (considered to be part of OpenCL)
>* OpenCL run-time, which runs on **A15**, but also on M4 (IPU1 which is reserved for TIDL OpenCL monitor role)
>* TIDL API host wrapper, user space library

* Another use case of the A15 is to use it for maximum performance. TIDL supports network layer distribution among the (4 EVEs + 2DSPs) accelerators present in the AM572x processor. Among the recommended settings, it is mentioned: 
>  _SoftMax layer_ is terminal layer it is advised to do SoftMax either on **A15** (in user space) or using DSP.

aditya-gumparthi, 2021-07-30:
hey @reneherrero did you find any more resources? c++ ? i'm in the same place today 😄

reneherrero, 2021-07-30:
This seems to be the only option: https://www.ti.com/tool/PROCESSOR-SDK-AM57X