beagleboard/beaglebone-ai #35: BeagleBone AI SPI Enable

Zeki411, 2020-02-05:
Hello, Everyone !
i am so brand new to BeagleBone AI (AM5729) and now i am trying to enable SPI on it. I have tried turn it on by adding some following line in uEnv.txt but it didn't work:

cmdline=coherent_pool=1M net.ifnames=0 quiet cape_universal=enable
uboot_overlay_addr4=/lib/firmware/BB-SPIDEV0-00A0.dtbo
uboot_overlay_addr5=/lib/firmware/BB-SPIDEV1-00A0.dtbo

Can anyone help me, pls ?

Lander-liu, 2020-03-21:
I met the same problem about uart ,so anyone help us?

silver2row, 2021-02-22:
Hello,

I think, and I am not affiliated at all except that I use their products, making a .dts file is a starter and then turning that into a .dtbo w/ the help of their scripts could be beneficial.

Seth

silver2row, 2021-03-18:
Hello Again,

Try here: https://groups.google.com/g/beagleboard/c/NFMZiBIPvno/m/2t_Oczx-AQAJ

Seth

closed 2021-07-01 by Zeki411