beagleboard/beaglebone-ai #37: How to enable uart 1 on the beaglebone AI

Lander-liu, 2020-03-23:
here is the BB-UART1-00A0 Decompile file， I find the regester has some problem, so anyone can help?

/dts-v1/;

/ {
	compatible = "ti,beaglebone", "ti,beaglebone-black", "ti,beaglebone-green";
	part-number = "BB-UART1";
	version = "00A0";
	exclusive-use = "P9.24", "P9.26", "uart1";

	fragment@0 {
		target-path = [2f 00];

		__overlay__ {

			chosen {

				overlays {
					BB-UART1-00A0 = "Thu Aug  1 16:09:14 2019";
				};
			};
		};
	};

	fragment@1 {
		target = <0xffffffff>;

		__overlay__ {

			P9_24_pinmux {
				status = "enabled";
			};

			P9_26_pinmux {
				status = "enabled";
			};
		};
	};
/*
	fragment@1 {
		target = <0xffffffff>;

		__overlay__ {

			P9_24_pinmux {
				status = "disabled";
			};

			P9_26_pinmux {
				status = "disabled";
			};
		};
	};
*/
	fragment@2 {
		target = <0xffffffff>;

		__overlay__ {

			pinmux_bb_uart1_pins {
				pinctrl-single,pins = <0x184 0x20 0x180 0x20>;
				linux,phandle = <0x1>;
				phandle = <0x1>;
			};
		};
	};

/*
	fragment@2 {
		target = <0xffffffff>;

		__overlay__ {

			pinmux_bb_uart1_pins {
				pinctrl-single,pins = <0x184 0x8 0x180 0x28>;
				linux,phandle = <0x1>;
				phandle = <0x1>;
			};
		};
	};

*/

	fragment@3 {
		target = <0xffffffff>;

		__overlay__ {
			status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <0x1>;
		};
	};

	__symbols__ {
		bb_uart1_pins = "/fragment@2/__overlay__/pinmux_bb_uart1_pins";
	};

	__fixups__ {
		ocp = "/fragment@1:target:0";
		am33xx_pinmux = "/fragment@2:target:0";
		uart1 = "/fragment@3:target:0";
	};

	__local_fixups__ {

		fragment@3 {

			__overlay__ {
				pinctrl-0 = <0x0>;
			};
		};
	};
};

iahn0, 2020-04-24:
I think uart1 is dedicated to console for beagleboneAI. I had no problem for console. We were able to FDDI usb converter to debug slot(3pins). 
  
If you are interested in, I was able to uart3  ttyS2 ( P9.22, P9.21) by adding the following part into custom.dtsam5729-beagleboneai-custom.dts . I also updated am5729-beagleboneai.dts file and compile to dtb files.  Copied to /boot/dtbs/ 
And you also need to clean up the second pins by deactivating pins.

Please refer to the resource the following link from element 14. It is very  helpful.  
 I have used the basic .dts file from the following link. https://www.element14.com/community/community/project14/visionthing/blog/2019/11/16/beagleboard-ai-brick-recovery-procedure 
 
This is my own way so you may want to change as you would like to do.  

//Beginning section, defining pin header on external port
#define P9_21 (0x3400 + 4 * 241) // PID is 241 //0x37C4
#define P9_22a (0x3400 + 4 * 167) // PID is 167 //unused //0x369C 
#define P9_22b (0x3400 + 4 * 240) // PID is 241 //0x37C0
//paste the following part 
&uart3 {
 status = "okay";
 pinctrl-names = "default";
 pinctrl-0 = <&uart3_pins>;
};
&dra7_pmx_core {
 uart3_pins: uart3 {
  pinctrl-single,pins = <
   DRA7XX_CORE_IOPAD( P9_21, MUX_MODE1 ) // P9.21 uart3_txd
   DRA7XX_CORE_IOPAD( P9_22b, PIN_INPUT_PULLUP | MUX_MODE1) // P9.22b uart3_rxd
   DRA7XX_CORE_IOPAD( P9_22a, MUX_MODE15) // P9.22a driver_off //shared-pin
  >;
 };
};

Insuk

p.s. I also shared my tests and information on 
https://groups.google.com/forum/embed/?&parenturl=https%3A%2F%2Fbeagleboard.org%2Fai&showsearch=true&showpopout=true&showtabs=true&place=forum%2Fbeagleboard&afterlogin&pli=1#!category-topic/beagleboard/beaglebone-ai/q5YLfysf_yU.

iahn0, 2020-04-24:
From schematic diagram, it looks like P9.24 and P9.26 are belong to uart10 group.
And Header P9.26 pin is sharing with ball number E21 ( need to  set mode to uart10 rxd  ) and ball number AE2 (need to set mode 15 for deactivating it instead of gpio mode, I think ). 

Physical address UART10 is 0x4AE2 B000.
In my beagle-AI, it was mapped to:  ttyS3 (aka ttyO3)
4ae2b000.serial/tty/ttyS3

I hope it is helpful.

paulwuertz, 2021-08-24:
> hi, I have the same problem. I need to use Uart10 as ttyS1. Please give me some ideas, what shoule i do.

I am quite a dummy for reloading device trees for pinmuxing myself, but I just made UART10 on pins P9.24 and P9.26a work ^^
Following this guide[1] and finding this overlay[2] on github I came up with this [file](https://github.com/beagleboard/beaglebone-ai/files/7042442/am5729-beagleboneai-custom.txt)  (should end with .dtb, but github only excepts .txt :P )

Follwing these steps from[1]:

```bash
cd ~ #(or wherever you want to clone the next line to)  
git clone https://github.com/beagleboard/BeagleBoard-DeviceTrees -b v4.14.x-ti  
nano BeagleBoard-DeviceTrees/src/arm/am5729-beagleboneai-custom.dts #(paste the dts file below)  
cd ../../BeagleBoard-DeviceTrees  
make src/arm/am5729-beagleboneai-custom.dtb  
sudo cp src/arm/am5729-beagleboneai-custom.dtb /boot/dtbs  
sudo nano /boot/uEnv.txt #(configure: dtb=am5729-beagleboneai-custom.dtb)  
sudo reboot  
```

where the `am5729-beagleboneai-custom.dts` is the file mentioned above. 

Before reboot I looked at all devices under `/dev/tty`. After reboot there was one more `/dev/ttyO1`, and when I did `echo foobar > /dev/ttyO1` I saw some 9600,8N,1 communication on the line ^^

[1] https://www.element14.com/community/community/project14/visionthing/blog/2019/11/16/beagleboard-ai-brick-recovery-procedure#jive_content_id_I2C_USE
[2] https://github.com/MalavPatel3501/BeagleBone-AI--mpBBAI-IO-Python-Library/blob/master/am5729-beagleboneai-mpcustom.dts