beagleboard/beaglebone-ai #38: HW: Possible cost reduction/simplification with single chip USB3 mux

jadonk, 2020-04-23:
Use TI HD3SS3220 instead of U22 (TI TUSB322IRWBR) and U72 (Pericom PI3USB3102ZLE) to save on size, cost, and complexity.

jadonk, 2020-05-14:
Rev A1:
[image]
[image]

Proposed Rev A2:
[image]

PrasanthElavarthi, 2020-07-26:
Are pin 12 and pin 22 of HD3SS3220 left floating? I cannot find a netlabel 3220_ID other than pin 12.