beagleboard/beaglebone-ai #44: Editor Eagle

kissup77, 2020-05-28:
Using latest eagle SW, I had opened files but it shows error "Start tag  ..."

kissup77, 2020-05-28:
I tried to open it with the latest EAGLE software

pdp7, 2020-05-28:
What files are you attempting to open in EAGLE

The BeagleBone AI was designed in Cadance Allegro.

BeagleBoard.org Foundation has contracted KiCad Professional Services to convert the design to KiCad, which is open source CAD software.

kissup77, 2020-05-28:
OMG, I thought it was designed with EAGLE. Let me try to open with Cadence Allegro

kissup77, 2020-05-28:
Actually I will need x,y coordinates to make samples of it. Can you share it with me?

pdp7, 2020-05-28:
@kissup77 do you mean centroid coordinates for a Pick n Place machine?

@jadonk do we have that information?

jadonk, 2020-06-03:
@pdp7 It should be part of the generated Gerbers, which we need to clean up. I might be able to regenerate it. Ideally, we'd verify against the production files.

kissup77, 2020-06-03:
One of Cadence Expert provides me x,y coordinates information files. It is resolved. Thanks.

closed 2020-06-03 by kissup77