beagleboard/beaglebone-ai #47: Which program should I use to open artwork files?

kissup77, 2020-07-08:
https://github.com/beagleboard/beaglebone-ai/tree/master/HW
I want to open the Artwork file.
I tried to open it with Altium, Pads, Eagle, but it failed.
Which program should I use to open artwork files?

pdp7, 2020-07-08:
Unfortunately, the design files are Cadence Allegro.  I believe there is a viewer here for Windows users: https://www.cadence.com/en_US/home/tools/allegro-downloads-start.html

However, we are currently having the design converted to KiCad.  We will announce when that is ready and add it to the repo.  And our goal is to use KiCad for future BeagleBoard.org designs.

pdp7, 2020-07-08:
If you have a more immediate need, please email me drew@beagleboard.org and I can see what the status is on the kicad conversion

closed 2020-07-09 by kissup77