beagleboard/beaglebone-ai #48: uart debugger not working

kissup77, 2020-07-09:
[image]
I attached the debugger as shown, but I can't see it.
The firmware used "AM5729 Debian 10.3 2020-04-06 16GB eMMC IoT TIDL Flasher".
The baud rate is 115200.
[image]
Is there something wrong with us?

DCelectronics, 2020-07-13:
It appears to me that you have correctly connected the GND, TX and RX, and the baud rate is correct (I just checked the one I have on my desk).

I suggest you use a regular image (not a flasher image, unless you intent to wipe your eMMC and re-install Debian)
https://debian.beagleboard.org/images/am57xx-debian-10.3-iot-tidl-armhf-2020-04-06-6gb.img.xz

Also, I suggest you check for shorts where the wires were soldered on the board (note that there is a tiny resistor between RX and GND pin on the board) and make sure that you have selected the correct COM port (I tend to have more the one USB-UART adaptors on my computer).

This cable would also make UART connections easier (I am posting this because this cable was not easy to find it).
https://www.newark.com/element14/1103004000156/beaglebone-ai-serials-cable/dp/50AH3702?MER=sy-me-pd-mi-acce

kissup77, 2020-07-13:
There is no short between RX and GND. I had checked with Voltmeter. No short.
So, It looks like CPU does not operate properly. Are there any point to check in voltage level? OSD should bring out 3.3V in VDD_3V3 while it does not bring up that voltage. What should I look more? I suspect PGOOD() or PSENSE pin option since it does not look like operate/run properly.

[image]

kissup77, 2020-07-13:
There is no short between RX and GND. I had checked with Voltmeter. No short.So, It looks like CPU does not operate properly. Are there any point to check in voltage level? OSD should bring out 3.3V in VDD_3V3 while it does not bring up that voltage. What should I look more? I suspect PGOOD or PSENSE pin option since it does not look like operate/run properly.

DCelectronics, 2020-07-14:
How do you power your Beaglebone AI board?
What power supply do you use (wall-wart, powered USB hub etc), and what is its current rating?

You can get the Beaglebone AI schematic from here:
https://github.com/beagleboard/beaglebone-ai/raw/master/BeagleBone-AI_sch.pdf
Go to page 14 of 19 (section D2) and look at the connector P9.
GND pins 1, 2
VDD_3V3 pins 3, 4
VDD_CAPE_5V pins 5, 6
VDD_5V pins 7, 8
I suggest you check the voltages on these pins.

A side note, it has been a while since I last looked at the Beaglebone AI SRM, and the team has made lots of progress.
https://github.com/beagleboard/beaglebone-ai/wiki/System-Reference-Manual#connecting-a-3-pin-serial-debug-cable

silver2row, 2021-03-18:
Hello...is that a BBG?

Seth