beagleboard/beaglebone-ai #59: Device tree overlay for ov2640 camera module with Beaglebone AI

Anitha241991, 2021-05-08:
We want to use ov2640 camera module with Beaglebone AI for our project. We found that ov2640.ko.xz is available under in the Linux Buster version 10. kernel version 4.19.

We understand that we will have to make the dtbo file to make the kernel understand our pin connections. However, all literature online is related to $pins and $slots which is obsolete.

We want to write our device tree overlay and go the route where just editing the uEnv.txt is enough to make kernel pick the dtbo.

Can you please guide us with the proper steps as we are kind of lost here.

niyazFattahov, 2021-07-17:
How you gonna connect dvp camera to BBAI?
It does not have camera inputs,  only lcd output according to schematics

Anitha241991, 2021-07-17:
Okay thank you.